bigtreetech/BTT-Expansion-module #5: BTT RRF WIFI 1.0

bagobor, 2020-10-13:
M587 S"..." P"..." leads to Failed to add SSID to remembered list

bagobor, 2020-10-13:
After doing some (re)search, I've figured out that wifi module wasn't flashed with any firmware. 

The way it worked for me:
1. hold the button
2. holding the boot button connect the wifi board to your computer
3. execute flashing cmd/tool.

Also wifi pins are need to be set properly in board.txt configuration file. 

I'm grateful for use to connect board but very dissatisfied by total lack of any support/documentation.
Not user friendly.
All links are to some external projects and you need to dig deep to establish some background first in order to make it work. 

It may be due very small number of boards released but anyway it'll be nice to provide complete packages for atleast SKR 1.3/4 boards configured for most popular stepper drivers (ie 2208/2209/etc) and bed leveling solutions.

fredopato, 2020-12-29:
Can you please give further instructions what you canged in the board.txt? In my Case i cannot geht a connection, but when i look in board.txt it seems right.

`//Analogue to Digital prefilter
adc.prefilter.enable = true

	//ESP Settings
8266wifi.espDataReadyPin = 0.28;
8266wifi.lpcTfrReadyPin = 1.30;
8266wifi.espResetPin = 1.31;`

Thank you

lordbeavis, 2021-01-03:
> After doing some (re)search, I've figured out that wifi module wasn't flashed with any firmware.
>
> The way it worked for me:
>
> 1. hold the button
> 2. holding the boot button connect the wifi board to [...]

I am having issues flashing the adapter.
Do you hold both buttons then connect?
When do you release the buttons?
I haven't tried just the boot button yet. I will update this when I do

fredopato, 2021-01-03:
> I am having issues flashing the adapter.
> Do you hold both buttons then connect?
> When do you release the buttons?
> I haven't tried just the boot button yet. I will update this when I do

Connect the Board via usb to your Computer, hit the "Flash NodeMCU" button in NodeMCU Flasher and then press RST and Boot, then release the RST button. Fyi: it only worked for me when i checked "yes, wipe all data", but maybe you should trie it first without this option checked. 

Also: You have to check the right combination of RepRap Firmware, ESP Firmware (DuetWiFiServer)  and "www" folder contents!

lordbeavis, 2021-01-03:
> > > After doing some (re)search, I've figured out that wifi module wasn't flashed with any firmware.
> > > The way it worked for me:
> > > 
> > > 1. hold the button
> > > 2. holding the boot button connect the wifi board to your computer
> > > 3. execute flashing cmd/tool.
> > > 
> > > Also wifi pins are need to be set properly in board.txt configuration file.
> > > I'm grateful for use to connect board but very dissatisfied by total lack of any support/documentation.
> > > Not user friendly.
> > > All links are to some external projects and you need to dig deep to establish some background first in order to make it work.
> > > It may be due very small number of boards released but anyway it'll be nice to provide complete packages for atleast SKR 1.3/4 boards configured for most popular stepper drivers (ie 2208/2209/etc) and bed leveling solutions.
> > 
> > 
> > I am having issues flashing the adapter.
> > Do you hold both buttons then connect?
> > When do you release the buttons?
> > I haven't tried just the boot button yet. I will update this when I do
> 
> Connect the Board via usb to your Computer, hit the "Flash NodeMCU" button in NodeMCU Flasher and then press RST and Boot, then release the RST button. Now it should start flashing. At this point you can release the Boot button and wait for the flashing to end. After that disconnect USB and install the RRF Wifi in your Printer.
> 
> Fyi: it only worked for me when i checked "yes, wipe all data", even if all sources say otherwise. But maybe you should trie it first without this option checked.
> 
> Also: You have to check the right combination of RepRap Firmware, ESP Firmware (DuetWiFiServer) and "www" folder contents!

thanks, i found the issue. i had to plug it into a usb 2 port and then make windows 10 download the proper serial driver