bigtreetech/BTT-Expansion-module #6: Marlin tutorial for installation BTT EXP-MOT V1.0 with TMC2209?

jerbotofu, 2020-11-10:
Hi! im trying connect btt exp mot with tmc2209 for a 2 x carriage but give me a error, need a tx and rx pin? please have any tutorial? thanks

TheStankec, 2021-01-09:
Do you see on the bottom of the boards, you have pinout marked and on exp-mot look for 
Exp-mot   | M1RX<->1.21  |  SKR 1.4
                 | M2RX<->1.19  |
                 | M3RX<->0.28  |

TheStankec, 2021-01-09:
[image]

TheStankec, 2021-01-09:
[image]

TheStankec, 2021-01-09:
Its talking... but when i try to move motors they just buzz and nothing else. if anyone has idea what that is about.
oh stepper drivers are ok I tried them on skr board and stepper motor also works

TheStankec, 2021-01-09:
oh and pins txt so you can se what is diferent

/**
 * Marlin 3D Printer Firmware
 * Copyright (c) 2020 MarlinFirmware [https://github.com/MarlinFirmware/Marlin]
 *
 * Based on Sprinter and grbl.
 * Copyright (c) 2011 Camiel Gubbels / Erik van der Zalm
 *
 * This program is free software: you can redistribute it and/or modify
 * it under the terms of the GNU General Public License as published by
 * the Free Software Foundation, either version 3 of the License, or
 * (at your option) any later version.
 *
 * This program is distributed in the hope that it will be useful,
 * but WITHOUT ANY WARRANTY; without even the implied warranty of
 * MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
 * GNU General Public License for more details.
 *
 * You should have received a copy of the GNU General Public License
 * along with this program.  If not, see <https://www.gnu.org/licenses/>.
 *
 */
#pragma once

#ifndef BOARD_INFO_NAME
  #define BOARD_INFO_NAME "BTT SKR V1.4"
#endif

//
// SD Connection
//
#ifndef SDCARD_CONNECTION
  #define SDCARD_CONNECTION                  LCD
#endif

//
// Servos
//
#define SERVO0_PIN                         P2_00

//
// TMC StallGuard DIAG pins
//
#define X_DIAG_PIN                         P1_29  // X-STOP
#define Y_DIAG_PIN                         P1_28  // Y-STOP
#define Z_DIAG_PIN                         P1_27  // Z-STOP
#define Z2_DIAG_PIN                         P1_20  // Z-STOP
#define Z3_DIAG_PIN                         P1_18  // Z-STOP
#define Z4_DIAG_PIN                         P1_30  // Z-STOP
#define E0_DIAG_PIN                        P1_26  // E0DET
#define E1_DIAG_PIN                        P1_25  // E1DET

//
// Limit Switches
//
#ifdef X_STALL_SENSITIVITY
  #define X_STOP_PIN                  X_DIAG_PIN
  #if X_HOME_DIR < 0
    #define X_MAX_PIN                      P1_26  // E0DET
  #else
    #define X_MIN_PIN                      P1_26  // E0DET
  #endif
#else
  #define X_STOP_PIN                       P1_29  // X-STOP
#endif

#ifdef Y_STALL_SENSITIVITY
  #define Y_STOP_PIN                  Y_DIAG_PIN
  #if Y_HOME_DIR < 0
[... 364 lines cut ...]
#endif // HAS_SPI_LCD

#if HAS_ADC_BUTTONS
  #error "ADC BUTTONS do not work unmodifed on SKR 1.4, The ADC ports cannot take more than 3.3v."
#endif

//
// NeoPixel LED
//
#ifndef NEOPIXEL_PIN
  #define NEOPIXEL_PIN                     P1_24
#endif

/**
 * Special pins
 *   P1_30  (37) (NOT 5V tolerant)
 *   P1_31  (49) (NOT 5V tolerant)
 *   P0_27  (57) (Open collector)
 *   P0_28  (58) (Open collector)
 */

//
// Include common SKR pins
//
#include "pins_BTT_SKR_common.h"

mindfreak269, 2021-04-06:
@TheStankec do you have found solustion ? i have same problem with you, driver ok, stepper can lock (enabled) but wont move.

JJare, 2021-04-06:
the same problem here, maybe a conflict with LCD or SD support??

Gruenbacher, 2021-04-29:
Hi JJare, 
i have the same problem and find a workaround for me.
In the file src/HAL/LPC1768/spi_pins.h 

original:
/** onboard SD card */
//#define SCK_PIN           P0_07
//#define MISO_PIN          P0_08
//#define MOSI_PIN          P0_09
//#define SS_PIN            P0_06
/** external */
#ifndef SCK_PIN
  #define SCK_PIN           P0_15
#endif
#ifndef MISO_PIN
  #define MISO_PIN          P0_17
#endif
#ifndef MOSI_PIN
  #define MOSI_PIN          P0_18
#endif
#ifndef SS_PIN
  #define SS_PIN            P1_23
#endif
#if !defined(SDSS) || SDSS == P_NC // gets defaulted in pins.h
  #undef SDSS
  #define SDSS              SS_PIN
#endif

my version:
/** onboard SD card */
#define SCK_PIN           P0_07
#define MISO_PIN          P0_08
#define MOSI_PIN          P0_09
#define SS_PIN            P0_06
/** external */
#ifndef SCK_PIN
  #define SCK_PIN           P0_15
#endif
#ifndef MISO_PIN
  #define MISO_PIN          P0_17
#endif
#ifndef MOSI_PIN
  #define MOSI_PIN          P0_18
#endif
#ifndef SS_PIN
  #define SS_PIN            P1_23
#endif
#if !defined(SDSS) || SDSS == P_NC // gets defaulted in pins.h
  #undef SDSS
  #define SDSS              SS_PIN
#endif

If the MOSI_PIN is not previously defined, then the compiler take in the original code the MOSI_PIN from P0_18 (on the EXP-MOT there is first motor step pin) and the MISO_PIN is used from P0_17 that is used from the third motor on the EXP-MOT Board., and the second motor is running normal. 

I hope this will help you.

Sven

cmdtbalbosa, 2021-05-09:
> ‎o mesmo problema aqui, talvez um conflito com suporte a LCD ou SD?? Apenas Motor2 funcionando corretamente, motor1 zumbido e motor3 não funcionando‎

I'm with a similar problem I put the Z2 in the E1 but when I trigger the engine it is buzzing (signal that this with high amperage) only I below the amperage and continues buzzing, I tried to do by marlin, in TFT, in trimport but to no avail. if I find out anything I share

sjporte93, 2021-06-22:
> Hi JJare,
> i have the same problem and find a workaround for me.
> In the file src/HAL/LPC1768/spi_pins.h
>
> original:
> /** onboard SD card _/ //#define SCK_PIN P0_07 //#define MISO_PIN P0_08 //#define [...]

I hit the same issue and although the above didn't fix it for me it did lead me down the right path, in configuration_adv.h I had SDCARD_CONNECTION connection set to LCD not sure if this is default, but that was causing my extra steppers to not move, so simply setting it to be "#define SDCARD_CONNECTION ONBOARD" then corrected the issue. and if you are interested where the values are being set it's within the  Marlin\src\pins\lpc1768\pins_BTT_SKR_common.h line 131 ish. essentially setting it to be the same as Gruenbacher, but in a cleaner way that avoids conflicts when building (which I had) 

hopefully that additional info is useful :)

I do however seem to have another issue where I can only build if I have one serial_port specified in configuration.h, I know it's another issue for another thread, but curious if anyones experienced the same with this btt setup?

cmdtbalbosa, 2021-07-14:
I did this setup but I'm having a little problem.
the application would be to use the M1 for Z2
M2 to Z3
the M1 is working properly but the M2 does not work the engine is as if it were with the wires reversed (that is not the case)

I've tested everything and just missing a detail that I imagine may be the problem.

what value did you put in config.h in Serial port 1 and serial port 2?