bigtreetech/BTT-Expansion-module #11: EEPROM install

Cabinfourus, 2021-01-17:
I have been trying to install the external EEPROM for more than two months. I keep getting the error for data size mismatch. I have tried different versions of Marlin, along with different configs. 
SKR 1.4 Turbo with TMC 2209 sensorless homing and 3D Touch sensor.
I have tried adding #define I2C_EEPROM to the pins file without any luck.

sdavos, 2021-01-23:
I have the same problem, unable to make eeprom working

fc1967, 2021-01-31:
The same for me, I have installed the external BTT EEPROM and tried many different configs in Marlin but I always keep the same error. I have a SKR 1.4 Turbo with TMC 2209, 3D Touch sensor and Smart Sensor.
I have tried adding #define I2C_EEPROM and also MARLIN_EEPROM_SIZE to the pins file and config file without luck.

MrFielding, 2021-02-02:
Same for me,anybody find a solution???tried with any tutorial but nothing worked.

fc1967, 2021-02-02:
I've also tried with two different boards but nothing, the same problem. there is any way to contact the technical assistance?

tgkgate, 2021-02-18:
Same issue,  though;  I use endstops and a genuine bltouch w/ 2209's.   I'm also using power_control w/ dedicated 5v supplied to the board separately.

radek8, 2021-06-09:
Hello.
I use the EEPROM expansion module together with the SKR V1.4 board. I don't have a module from BBT but a clone from Aliexpress. But the use is the same.
In order for it to work, it is necessary to edit the pin layout file.
If anyone is interested, I can attach the necessary adjustments to make the module work.
The module is connected to the I2C connector
The first time you start, you need to initialize the EEPROM

// EEPROM I2C
#define I2C_EEPROM
#ifdef E2END
   #undef E2END
#endif
#define E2END 0x7FFF        // EEPROM end address 24C256 (32 kB = 8 kB)
//# define E2END 0xFFFF   // end address EEPROM 24C512 (64 kB = 8 kB)