bigtreetech/BTT-Expansion-module #14: BTT-EXP-Mot & SKR 1.4T Working!

BigDaddy4, 2021-03-16:
I finally was able to get an SKR 1.4T & Exp-Mot working together.
My Config is a CoreXY with 4 Z Motors and The New BugFix 2.X Marlin.
Marlin 2.0.7.2 does not work with the exp-mot, unless you want to manually customize it yourself.
open Pins_BTT_SKR_common.h
Remove the "//" from the line //#define HAS_BTT_EXP_MOT 1
in Configuration_adv.h I changed #define NUM_Z_STEPPER_DRIVERS 2 to #define NUM_Z_STEPPER_DRIVERS 4
I had errors while compiling at this point "You can't use both an LCD and a Motor Expansion Module on EXP1/EXP2 at the same time."

I had to Disable "//#define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER
Then everything worked for me.
I hope this helps people.

mindfreak269, 2021-04-04:
@BigDaddy4 i want to ask about exp mot, i have skr 1.4 installed 5 tmc2209 uart, i want to expand 3 stepper for more extruder, i have configure all pin, but in exp mot i'm install a4988 in DIR mode, but only when use stepper in exp mot, it wont move, any solution for that? not much tutorial or explanation about exp mot in internet. thank you

wickytje, 2021-04-16:
@BigDaddy4   I am building blv mgn cube and wanted  to do 4 Z Motors thanks .

 what kind of display are you using