bigtreetech/BTT-Expansion-module #24: BTT MOT EXP V1.0 board - Second Motor active when 1st is in use KLIPPER

VORONENTHUSIAST, 2023-10-22:
So, I have a problem where the second motor on my BTT MOT EXP board is active whether or not it's in use. This is my 3rd extruder module, First extruder is on my main board, second extruder is Port 1 on the MOT EXP, etc. etc. 

I've put my firmware excerpt below:

#####################################################################
# 	Extruder Motors
#####################################################################

# The default configuaration for this firmware is the Bowden Feed, using the Mobius M4 extruder. 
# If you intend to use the Afterburner Direct Drive, please change the gear ratio to 50:17 and calculate your roation distance prior to calibrating.
# This firmware is configured to run a 3into1 extrusion setup. It utilises the Big Tree Tech MOT EXP v1.0 board to run 2 extra motors for the print head.
# PLEASE NOTE - If you are using another type of feeding system, you will need to edit this heavily.

# "Extruder" Motor - Use "E0" Port on MKS SKIPR V1.0 board

[extruder]                          
step_pin: PB5                       
dir_pin: PB4                        
enable_pin: !PB6                  
microsteps: 16
rotation_distance: 22.56	        # Bondtech 5mm Drive Gears
gear_ratio: 80:20                   # Mobius M4 Bowden 80:20
full_steps_per_rotation: 200	    # 200 for 1.8 degree, 400 for 0.9 degree
nozzle_diameter: 0.800
filament_diameter: 1.750
min_temp: 0
max_temp: 275
heater_pin: PB1
sensor_type: Generic 3950
sensor_pin: PC1
max_power: 0.7
control : pid  
pid_kp : 19.207
pid_ki : 0.865
pid_kd : 106.597
pressure_advance: 0.65              # Caluclated using the pa-cal macro. This needs to calculated for each individual filament you use.
pressure_advance_smooth_time: 0.150
max_extrude_cross_section: 10
instantaneous_corner_velocity: 1.000
max_extrude_only_distance: 650.0
max_extrude_only_velocity:5000
max_extrude_only_accel:2000
step_pulse_duration:0.000004

# "Second_Extruder" Motor - Use "M1" Port on BTT MOT EXP V1.0 board

[extruder_stepper second_extruder]
extruder: extruder                          
step_pin: PA7                       # EXP2_6
dir_pin: PE8                        # EXP2_5
enable_pin: !PD13                   # !EXP2_7
microsteps: 16
rotation_distance: 22.56	        # Bondtech 5mm Drive Gears
gear_ratio: 80:20                   # Mobius M4 Bowden 80:20
full_steps_per_rotation: 200	    # 200 for 1.8 degree, 400 for 0.9 degree
pressure_advance: 0.65              # Caluclated using the pa-cal macro. This needs to calculated for each individual filament you use.
pressure_advance_smooth_time: 0.150

# "Third_Extruder" Motor - Use "M2" Port on BTT EXP-MOT V1.0 board

[extruder_stepper third_extruder]  
extruder: extruder
step_pin: PE9                       # EXP2_3
dir_pin: !PA4                       # EXP2_4
enable_pin: !PE7                    # !EXP1_8
microsteps: 16
rotation_distance: 22.56            # Bondtech 5mm Drive Gears
full_steps_per_rotation: 200        # 200 for 1.8 degree, 400 for 0.9 degree
gear_ratio: 80:20                   # Mobius M4 Bowden 80:20
pressure_advance: 0.65              # Caluclated using the pa-cal macro. This needs to calculated for each individual filament you use.
pressure_advance_smooth_time: 0.150

# "Fourth_Extruder" Motor - Use "M3" Port on BTT EXP-MOT V1.0 board.

[extruder_stepper fourth_extruder]   
extruder: extruder
step_pin: PA6                        # EXP2_1
dir_pin: PA5                         # EXP2_2
enable_pin: !PE15                    # !EXP1_7
microsteps: 16
rotation_distance: 22.56             # Bondtech 5mm Drive Gears
full_steps_per_rotation: 200         # 200 for 1.8 degree, 400 for 0.9 degree
gear_ratio: 80:20                    # Mobius M4 Bowden 80:20
pressure_advance: 0.65               # Caluclated using the pa-cal macro. This needs to calculated for each individual filament you use.
pressure_advance_smooth_time: 0.150

I will say with this, I haven't changed the extruder setup in Prusa Slicer to include motors 3 and 4 yet, could this be the reason why?

VORONENTHUSIAST, 2023-10-24:
I worked it out - I forgot to change my macro when I added my extra extruders.

Shamoon78, 2024-01-16:
Hi Mate and what did you use for Uart CS pin ?

tanjunior, 2024-09-07:
> Hi Mate and what did you use for Uart CS pin ?

yea i want to know as well

VORONENTHUSIAST, 2024-09-08:
@Shamoon78 @tanjunior 

Apologies, I've been away from the hobby for a while - M1 Port uses PD8 (EXP1_6) M2 uses PD10 (EXP1_4) M3 uses PE10 (EXP1_2)

Hope that helps

closed 2024-09-08 by VORONENTHUSIAST

tanjunior, 2024-09-08:
> @Shamoon78 @tanjunior
> 
> Apologies, I've been away from the hobby for a while - M1 Port uses PD8 (EXP1_6) M2 uses PD10 (EXP1_4) M3 uses PE10 (EXP1_2)
> 
> Please note I'm using the MKS SKIPR V1.0 board, so this is matched up to that.
> 
> Hope that helps

I am also using EXP1_6 for my MKS ROBIN E3D, no idea why it's not working. My issue here #28 If you could help.

VORONENTHUSIAST, 2024-09-08:
I had to cut the tabs off of the ribbon cable connectors on the MKS SKIPR side and insert them the other way around, I can't see the pinout for the E3D board in the MKS github (coming up as an invalid .pdf) so can't confirm if you need to so the same 

> I am also using EXP1_6 for my MKS ROBIN E3D, no idea why it's not working. My issue here #28 If you could help.

I had to cut the tabs off of the ribbon cable connectors on the MKS SKIPR side and insert them the other way around, I can't see the pinout for the E3D board in the MKS github (coming up as an invalid .pdf) so can't confirm if you need to so the same