bigtreetech/BTT-Expansion-module #26: I am conncting Exp MOt v1.0 to octopus pro and using klipper

Shamoon78, 2024-01-16:
what to use as Uart pin for 3 motor drivers on the board ?is it same as rx pin ? which idid but not working .

Unable to read tmc uart 'extruder' register IFCNT
Once the underlying issue is corrected, use the
"FIRMWARE_RESTART" command to reset the firmware, reload the
config, and restart the host software.
Printer is shutdown
[image]
note :Using only M2 on the board as Extruder and the rest 8 motors are all used on octopus pro .

norris543, 2024-02-12:
Did you ever figure out a solution?

Shamoon78, 2024-02-13:
> Did you ever figure out a solution?

Yes fixed but iforget what was the problem , I think mostley i forget to remove the jumpers and leave only Uart mode one jumper and removing the other 3 . (changing from spi to uart mode

norris543, 2024-02-13:
do you mind sending a photo of how you have it setup? I am running an skr3
ez and yrying to figure out if its how I have my jumpers or if its because
I am using ez adapters on the expansion board.