bigtreetech/BTT-Expansion-module #28: BTT EXP-MOT with MKS Robin E3D

tanjunior, 2024-09-07:
I wanted to add 1 additional stepper driver to my MKS ROBIN E3D so that I can do Z_TILT.

E3D UART mode
MOT-EXP UART mode
E3D EXP1 <-> EXP-MOT EXP1
E3D EXP2 <-> EXP-MOT EXP2

Following the [BTT EXP-MOT user guide](https://github.com/bigtreetech/BTT-Expansion-module/blob/master/BTT%20EXP-MOT/BTT%20EXP-MOT%20V1.0User%20Guide.pdf) and cross referencing with the [E3D pins](https://github.com/makerbase-mks/MKS-Robin-E3-E3D/blob/master/hardware/MKS%20Robin%20E3D%20V1.1_003/MKS%20Robin%20E3D%20V1.1_003%20PIN.pdf)

BTT-MOT EXP2 M1STP = E3D EXP2 PB15
BTT-MOT EXP2 M1DIR = E3D EXP2 PB0
BTT-MOT EXP2 M1EN = E3D EXP2 PC10
BTT-MOT EXP1 M1RX = E3D EXP1 PA7

```
[stepper_z1]
step_pin: EXP2_6
dir_pin: !EXP2_5
enable_pin: !EXP2_7
microsteps: 16
rotation_distance: 8
endstop_pin: ^PC6
endstop_pin: probe:z_virtual_endstop

[tmc2209 stepper_z1]
uart_pin: EXP1_6
run_current: 0.8
stealthchop_threshold: 999999

[board_pins]
aliases:
    # EXP1 header
    EXP1_1=PC1, EXP1_3=PA4, EXP1_5=PA6, EXP1_7=PC4, EXP1_9=<GND>,
    EXP1_2=PC3, EXP1_4=PA5, EXP1_6=PA7, EXP1_8=PC5, EXP1_10=<5V>,
    # EXP2 header
    EXP2_1=PB14, EXP2_3=PB11, EXP2_5=PB0,  EXP2_7=PC10,  EXP2_9=<GND>,
    EXP2_2=PB13, EXP2_4=PA15, EXP2_6=PB15, EXP2_8=<RST>, EXP2_10=<NC>,
    # CR_EXP3 header - additional header dedicated to Creality Ender 3/5 stock screens
    CR_EXP3_10=<5V>, CR_EXP3_8=PA4, CR_EXP3_6=PA6, CR_EXP3_4=<RST>, CR_EXP3_2=PC3,
    CR_EXP3_9=<GND>, CR_EXP3_7=PA5, CR_EXP3_5=PB0, CR_EXP3_3=PB11,  CR_EXP3_1=PC1
```

`FORCE_MOVE STEPPER=stepper_z1 DISTANCE=10 VELOCITY=5`
[image]

VORONENTHUSIAST, 2024-09-08:
Can you show how your plugs are inserted into both boards, please?

Looking at the pinout you've shown against the pinout for the BTT EXP-MOT, they are reversed. 
Pin 1 is is bottom left for the BTT board when looking at it directly, Pin 1 is top right on your MKS board - this is the same as my SKIPR Board. 

Other than that all I can think is maybe not use aliases? It seems much more logical to just use the pin names from your MKS E3D and hash the EXP number next to it.

[image]
[image]

That's extruder code taken from the working file of one of my Voron 2.4s.

TLDR: BTT and MKS EXP ports are reversed, the cables of one side (I chose mainboard) must have their tabs cut off and be rotated 180 degrees. If you already did this, just try not using aliases and see what happens?

tanjunior, 2024-09-08:
> Can you show how your plugs are inserted into both boards, please?

[image]

> the cables of one side (I chose mainboard) must have their tabs cut off and be rotated 180 degrees.

the port on my MKS side does not have a housing surrounding it, allows me to connect in either orientation.

>try not using aliases and see what happens?

I will try and give you an update

tanjunior, 2024-09-09:
@VORONENTHUSIAST Still getting the same error without using alias.

VORONENTHUSIAST, 2024-09-09:
> @VORONENTHUSIAST Still getting the same error without using alias.

I'm all out of ideas then I'm afraid.
Mine is up and running using the EXP ports I can see in your shared config.

Did you ever plug it in the wrong way around? I'm just wondering if you've popped the board perhaps. 
All wires are plugged in for both EXP ports?

tanjunior, 2024-09-09:
> Did you ever plug it in the wrong way around? I'm just wondering if you've popped the board perhaps.

What do you meant by popped?

> All wires are plugged in for both EXP ports?

yes both are plugged in!

Also wondering if defining an extra MCU is needed

VORONENTHUSIAST, 2024-09-09:
By popped, I mean damaged.

No defining of an MCU is needed - the BTT EXP-MOT doesn't have a control chip like your main board

tanjunior, 2024-09-09:
> By popped, I mean damaged.

I notice that the Z2 stepper jerks when I power on my printer. Other than that no smell and onboard LEDS are on.

VORONENTHUSIAST, 2024-09-09:
Are you using all 3 ports on the board, if so are the other 2 working?

tanjunior, 2024-09-09:
> Are you using all 3 ports on the board, if so are the other 2 working?

I only need 1 I have only tested with M1 so far. Now that you mentioned it I should had tested the other 2

tanjunior, 2024-09-15:
```
[stepper_z1] #m1
step_pin: PB15 #EXP2_6
dir_pin: !PB0 #!EXP2_5
enable_pin: !PC10 #!EXP2_7
microsteps: 16
rotation_distance: 8
endstop_pin: ^PC6
endstop_pin: probe:z_virtual_endstop

[stepper_z2] #m2
step_pin: EXP2_3
dir_pin: !EXP2_4
enable_pin: !EXP1_8
microsteps: 16
rotation_distance: 8
endstop_pin: ^PC6
endstop_pin: probe:z_virtual_endstop

[stepper_z3] #m3
step_pin: EXP2_1
dir_pin: !EXP2_2
enable_pin: !EXP1_7
microsteps: 16
rotation_distance: 8
endstop_pin: ^PC6
endstop_pin: probe:z_virtual_endstop

[tmc2209 stepper_z1] #m1
uart_pin: PA7 #EXP1_6
# diag_pin: PA6 #EXP1_5
run_current: 0.8
stealthchop_threshold: 999999

[tmc2209 stepper_z2] #m2
uart_pin: EXP1_4
# diag_pin: EXP1_3
run_current: 0.8
stealthchop_threshold: 999999

[tmc2209 stepper_z3] #m3
uart_pin: EXP1_2
# diag_pin: EXP1_1
run_current: 0.8
stealthchop_threshold: 999999
```

I have tried all 3 and it's not working.

VORONENTHUSIAST, 2024-10-14:
No clue then unfortunately, as I said, mine is working

kaswets, 2026-01-07:
I have a octopus Pro board and a exp-mot.

can I run Uart mode on the Exp-mot?  and what pins need I to select in klipper printer.cfg?

Please help!